rkoptev/ACS712-arduino #9: current measurement by acs sensor is unstable

fardah97, 2018-07-26:
hello, I have some trouble measuring current using ACS712 and NodeMCU ESP8266, there are:

1. I have ACS712 30A and 5 watt lamp, when I use ACS712_30A the result shown is 30 watts without load. but when I use ACS712_05B the result shown is 5.81 watts. Why does this happen?

2. when I keep using ACS712_05B I get initially a constant result. when there is no current flow, the sensor reads 0 watts, when there is a current flow, sensor reading 5.81 watts. but when I try many times the calculations become more unstable. when there is no current, the sensor reads 5.81 watts up to 10 watts. I can not solve this problem

note : I use relay to control the lamp using mobile app. I also use Vin Pin on NodeMCU instead of 3.3v

[ACSmobile.zip](https://github.com/rkoptev/ACS712-arduino/files/2230307/ACSmobile.zip)

rkoptev, 2018-07-26:
Hi @fardah97 ,

First, how do you connect the sensor to the ESP8266? Could you provide a connection diagram?

Running ahead I will say that it is also very important to have a good power source, the voltage of which is constant and does not drop significantly under the load from the relay.

fardah97, 2018-07-28:
[image]

rkoptev, 2018-07-30:
Thank you for the wiring diagram, @fardah97 .
It looks like you are powering sensor and relays from one power source, and as I understand it is USB power. That may bring to power drops when relay's switching on and off, which will cause to slightly distorted readings. My advice is to try to use one power source for ESP and sensor and separate one for relays and other stuff that can influence power source. For instance, you can power ESP and sensor from USB and take 5V wall adapter for other stuff.