rkoptev/ACS712-arduino #10: Ampere Value returned minus number

hammeractual, 2018-07-27:
Hi there.
I'm using your lib and use the class of getCurrentDC() in MeasureCurrentDC example, but why in the serial monitor the amp value retured - (minus) numbers?

rkoptev, 2018-07-27:
This is because ACS712 is bidirectional current sensor. It means it can measure both direct and reverse current. To make your values positive you can change two load wires that you connected to sensor.

hammeractual, 2018-07-27:
>  change two load wires that you connected to sensor
How do I do it, Sir?

rkoptev, 2018-07-27:
I mean you need to swap two wires that goes to bigger connector of sensor.

hammeractual, 2018-07-27:
But sometimes the value returned negatives and sometimes it become postives even i don't swap the wires.

rkoptev, 2018-07-30:
Similar problem has already been discussed here: https://github.com/rkoptev/ACS712-arduino/issues/9