rkoptev/ACS712-arduino #12: Error while compiling

JeffrySteegmans, 2018-09-09:
I get the following error when compiling with platformIO

```
/lib/ACS712 Current Sensor_ID1601/src/ACS712.cpp:46:51: error: default argument given for parameter 1 of 'float ACS712::getCurrentAC(uint16_t)' [-fpermissive]
```

The problem is solved when take away the default argument for method ACS712::getCurrentAC

```
float ACS712::getCurrentAC(uint16_t frequency)
```

JeffrySteegmans, 2018-09-11:
Hi rkoptev,

I get the error if I build with the latest version of platformIO with visual studio Code
[https://pastebin.com/raw/bqXyCK5C](url)

The error also occures when building with the Travis CI framework:
[https://travis-ci.org/JeffrySteegmans/ESP-Powerswitch](url)
[https://pastebin.com/raw/qX2EwQ8G](url)

You can see my code at github:
[https://github.com/JeffrySteegmans/ESP-Powerswitch](url)

rkoptev, 2018-09-11:
I figured out the issue. Looks like I forgot to increment a release number during the last commits, so Platformio not grabbed latest changes. I updated properties file to newer version (1.0.2). Platformio must parse it in near time. For now you can wait while Platformio prompt you to update the library in the library manager, or you can just leave your modification to the library if you need to continue work ASAP.

Thank you for the reporting!

JeffrySteegmans, 2018-09-11:
Tx for the quick solution ;)

closed 2018-09-11 by JeffrySteegmans