rkoptev/ACS712-arduino #16: library unestable , it shows current when the lamp there is no lamp( open circuit)

martindipa, 2020-05-17:
I try the library using ACS712 5A,
 I am using arduino uno, a relay to turn on a 220v Led Lamp., Relay and ACS712 where in serial connection.
Issue description:  When there is no current flowing the ACS712, the values returned from I=sensor.getCurrentAC(); not changes,
When the lamp is ON with the relay in ON state, current flows thught ACS712 values are returned from I=sensor.getCurrentAC() , then I remove the lamp  so there is no posibilities to the current remains flowing, but that it is  not returned  in I=sensor.getCurrentAC().

Thanks 
Martin

djmallick, 2020-05-18:
ACS712 mainly works sensing the magnetic field created by AC. Now if you use a relay near the sensor, it will cause more magnetic field. Basically magnetic field due to AC is too weak compared to the relay magnetic field. So you will get too much higher value of reading due to this stray magnetic field when relay is ON.

SO best solution is - Try to make a magnetic shield around sensor. You can cover it using an Aluminium foil.  It will protect from stray magnetic field. Also should take care that foil doesn't cause any short circuit in the sensor.

martindipa, 2020-05-21:
Thanks for the reply, but the phenomen you describe not happens using another library(for ACS712 ) so the problem was the library. In both I use the same circuit, one work and ones not.

thanks

Bbadru, 2020-07-23:
> Thanks for the reply, but the phenomen you describe not happens using another library(for ACS712 ) so the problem was the library. In both I use the same circuit, one work and ones not.
> 
> thanks

can i see the other acs712 library